MEN-Mikro-Elektronik/13MD05-90 #249: documentation about Kernel headers under Debian is partial

dpfeuffer, 2022-11-18:
From MAIN_PR007880:
The documention for Ubuntu/Debian says :
____________________________________________________________________________________
Ubuntu / Debian

    Install the build toolchain:

    # apt-get install build-essential

    Install the kernel headers

        for specified kernel version:

        # apt-get install linux-headers-<kernel-version>

        for running kernel version:

        # apt-get install linux-headers-$(uname -r)

    Create the Symlink to the kernel headers

        for specified kernel version:

        # ln -s /usr/src/linux-headers-<kernel-version> /usr/src/linux

        for running kernel version:

        # ln -s /usr/src/linux-headers-$(uname -r) /usr/src/linux
______________________________________________________________________________

This is correct for Ubuntu, but not for Debian (Buster 10.11 at least).

I f you do "apt-get install linux-headers-$(uname -r)", the build of MDIS is not possible.
you have to do:

apt-get install linux-headers-4.19.0-18-common

to install the right headers that can be usable with MDIS.
see attachment.
[image]

dpfeuffer, 2022-11-18:
see also #246

mad-jsanjuan, 2022-11-22:
Looking at a Debian 10 distro. The Debian package manager will handle `apt-get install linux-headers-$(uname -r)` correctly and install the linux headers ready to be used for building loaddabel modules into the kernel:

```
men@debian:/usr/src$ apt show linux-headers-$(uname -r)
Package: linux-headers-4.19.0-22-amd64
Version: 4.19.260-1
Priority: optional
Section: kernel
Source: linux
Maintainer: Debian Kernel Team <debian-kernel@lists.debian.org>
Installed-Size: 5,342 kB
Depends: linux-headers-4.19.0-22-common (= 4.19.260-1), linux-kbuild-4.19 (>= 4.19.260-1), linux-compiler-gcc-8-x86
Homepage: https://www.kernel.org/
Download-Size: 1,107 kB
APT-Manual-Installed: yes
APT-Sources: http://security.debian.org/debian-security buster/updates/main amd64 Packages
Description: Header files for Linux 4.19.0-22-amd64
 This package provides the architecture-specific kernel header files for
 Linux kernel 4.19.0-22-amd64, generally used for building out-of-tree
 kernel modules.  These files are going to be installed into
 /usr/src/linux-headers-4.19.0-22-amd64, and can be used for building
 modules that load into the kernel provided by the
 linux-image-4.19.0-22-amd64 package.
men@debian:/usr/src$ ls -l /usr/src/
total 8
lrwxrwxrwx 1 root root   38 Nov 10 11:56 linux -> /usr/src/linux-headers-4.19.0-22-amd64
drwxr-xr-x 4 root root 4096 Nov 10 11:21 linux-headers-4.19.0-22-amd64
drwxr-xr-x 4 root root 4096 Nov 10 12:28 linux-headers-4.19.0-22-common
lrwxrwxrwx 1 root root   24 Sep 29 02:47 linux-kbuild-4.19 -> ../lib/linux-kbuild-4.19
```

If the symlink is created with `ln -s /usr/src/linux-headers-$(uname -r) /usr/src/linux` then all the setup of the kernel sources is correct as described in the documentation. 

What seems to not be correct for Debian is to pick up a custom kernel version with `apt-get install linux-headers-<kernel-version>` because there is no packages with such name, they all have a suffix (-amd64 for arch dependent or -common for generic headers). This seems to also be inaccurate in Ubuntu where we get a -generic suffixed package for architecture dependent headers and configs and a package with no suffix for generic stuff (unconfigured). 

```
jorge@jorge-Latitude-5530:/usr/src$ apt show linux-headers-$(uname -r)
Package: linux-headers-5.15.0-53-generic
Version: 5.15.0-53.59
Priority: optional
Section: devel
Source: linux
Origin: Ubuntu
Maintainer: Ubuntu Kernel Team <kernel-team@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 24,6 MB
Provides: linux-headers, linux-headers-3.0
Depends: linux-headers-5.15.0-53, libc6 (>= 2.34), libelf1 (>= 0.142), libssl3 (>= 3.0.0~~alpha1), zlib1g (>= 1:1.2.3.3)
Download-Size: 2.899 kB
APT-Manual-Installed: no
APT-Sources: http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
Description: Linux kernel headers for version 5.15.0 on 64 bit x86 SMP
 This package provides kernel header files for version 5.15.0 on
 64 bit x86 SMP.
 .
 This is for sites that want the latest kernel headers.  Please read
 /usr/share/doc/linux-headers-5.15.0-53/debian.README.gz for details.

jorge@jorge-Latitude-5530:/usr/src$ apt show linux-headers-5.15.0-53
Package: linux-headers-5.15.0-53
Version: 5.15.0-53.59
Priority: optional
Section: devel
Source: linux
Origin: Ubuntu
Maintainer: Ubuntu Kernel Team <kernel-team@lists.ubuntu.com>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 77,5 MB
Depends: coreutils
Download-Size: 12,3 MB
APT-Manual-Installed: no
APT-Sources: http://es.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
Description: Header files related to Linux kernel version 5.15.0
 This package provides kernel header files for version 5.15.0, for sites
 that want the latest kernel headers. Please read
 /usr/share/doc/linux-headers-5.15.0-53/debian.README.gz for details

```

mad-jsanjuan, 2022-11-29:
I found more info about this issue: It seems what gets confusing is some checks done by the MDIS Wizzard. This means, getting the linux headers for Debian and doing the `/usr/src/linux` symbolic link using the `uname -r` string is fine. The problem is when the MDIS wizzard tries to validate the linux sources at `/usr/src/linux`. It tries to find some files and because the Debian package splits the sources into two directories, it cannot find it and does not let the user create the MDIS project. 

[image]

closed 2023-01-12 by mad-jsanjuan