MEN-Mikro-Elektronik/13MD05-90 #252: Perform 13Z077-90 Pull Request

dpfeuffer, 2022-11-30:
Please perform this pull request
https://github.com/MEN-Mikro-Elektronik/13Z077-90/pull/10
and take the following into account:
- merge it into a new mad-dev branch but not directly into master
- functional testing is already done by us and a customer, so it is sufficient to perform compilation testing only
- take a look at the https://github.com/duagon-hhedwig/13Z077-90/tree/development branch and decide if it make sense to merge this changes (Checkpatch style improvements) also into the MEN-Mikro-Elektronik/13Z077-90 repo

mad-jsanjuan, 2022-12-01:
I tested these changes on this branch https://github.com/MEN-Mikro-Elektronik/13MD05-90/pull/253 and we have some compilation errors:
4.4.252_dbg FAILED
4.4.252_nodbg FAILED
4.9.252_dbg FAILED
4.9.252_nodbg FAILED
4.14.216_dbg FAILED
4.14.216_nodbg FAILED
4.19.168_dbg FAILED
4.19.168_nodbg PASSED
5.4.90_dbg FAILED
5.4.90_nodbg PASSED
5.6.19_dbg FAILED
5.6.19_nodbg PASSED
5.10.8_dbg FAILED
5.10.8_nodbg PASSED

It seems there is an issue with a debug message for the newer kernel versions in this function https://github.com/duagon-hhedwig/13Z077-90/blob/8ec4f12ad5d53ea9f7bd2770506b6bf3ed8cda81/DRIVERS/ETH_16Z077/DRIVER/men_16z077_eth.c#L2211: 

```
$ cat 5.10.8_dbg/Makefile.13Z077-90.log  | grep -C 5 error:
  CC [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon_io/chameleon_strings.o
  CC [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon_io/nat_module.o
  LD [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon_io/men_chameleon_io.o
  CC [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.o
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c: In function 'z77_poll':
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:2213:33: error: 'rx1' undeclared (first use in this function)
 2213 |                                 rx1, rx0, start_pos, nrframes );
      |                                 ^~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:133:63: note: in definition of macro 'Z77DBG'
  133 | #define Z77DBG(lvl,msg...) if (np->msg_enable >= lvl) printk( msg );
      |                                                               ^~~
--
 2213 |                                 rx1, rx0, start_pos, nrframes );
      |                                 ^~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:133:63: note: in definition of macro 'Z77DBG'
  133 | #define Z77DBG(lvl,msg...) if (np->msg_enable >= lvl) printk( msg );
      |                                                               ^~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:2213:38: error: 'rx0' undeclared (first use in this function)
 2213 |                                 rx1, rx0, start_pos, nrframes );
      |                                      ^~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:133:63: note: in definition of macro 'Z77DBG'
  133 | #define Z77DBG(lvl,msg...) if (np->msg_enable >= lvl) printk( msg );
      |                                                               ^~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:2213:43: error: 'start_pos' undeclared (first use in this function)
 2213 |                                 rx1, rx0, start_pos, nrframes );
      |                                           ^~~~~~~~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:133:63: note: in definition of macro 'Z77DBG'
  133 | #define Z77DBG(lvl,msg...) if (np->msg_enable >= lvl) printk( msg );
      |                                                               ^~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:2213:54: error: 'nrframes' undeclared (first use in this function)
 2213 |                                 rx1, rx0, start_pos, nrframes );
      |                                                      ^~~~~~~~
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_lx_z77/men_16z077_eth.c:133:63: note: in definition of macro 'Z77DBG'
  133 | #define Z77DBG(lvl,msg...) if (np->msg_enable >= lvl) printk( msg );
      |                                                               ^~~

```

There is also compilation issues for kernel versions older than `4.19.168` on the list: 

```
$ cat 4.14.216_nodbg/Makefile.13Z077-90.log  | grep -C 5 error:
  CC [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_chameleon/men_chameleon.o
  LD [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_chameleon/men_lx_chameleon.o
  AR      /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/built-in.o
  CC [M]  /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/men_16z077_eth.o
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/men_16z077_eth.c: In function 'z77_open':
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/men_16z077_eth.c:2273:54: error: 'HRTIMER_MODE_REL_PINNED_SOFT' undeclared (first use in this function); did you mean 'HRTIMER_MODE_REL_PINNED'?
 2273 |         hrtimer_init(&np->tx_timer, CLOCK_MONOTONIC, HRTIMER_MODE_REL_PINNED_SOFT);
      |                                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                      HRTIMER_MODE_REL_PINNED
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/men_16z077_eth.c:2273:54: note: each undeclared identifier is reported only once for each function it appears in
/home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/men_16z077_eth.c: In function 'z77_send_packet':
[... cut ...]
      |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                               HRTIMER_MODE_REL_PINNED
make[4]: *** [scripts/Makefile.build:327: /home/docker/mdis_test_system/MDIS_Compilation_Test/OBJ/nodbg/men_lx_z77/men_16z077_eth.o] Error 1
```

It seems the identifier HRTIMER_MODE_REL_PINNED_SOFT was first introduced in kernel version `4.16`:

https://elixir.bootlin.com/linux/v4.15/C/ident/HRTIMER_MODE_REL_PINNED_SOFT
https://elixir.bootlin.com/linux/v4.16/C/ident/HRTIMER_MODE_REL_PINNED_SOFT

mad-jrodriguez, 2023-01-02:
With the latest changes, all tests pass OK

```
4.9_dbg PASSED
4.9_nodbg PASSED
4.14_dbg PASSED
4.14_nodbg PASSED
4.19_dbg PASSED
4.19_nodbg PASSED
5.4_dbg PASSED
5.4_nodbg PASSED
5.10_dbg PASSED
5.10_nodbg PASSED
5.15_dbg PASSED
5.15_nodbg PASSED
```

A Pull Request has been created in subrepo 13Z077-90 to include such changes in mad-dev branch

You can track it here: https://github.com/MEN-Mikro-Elektronik/13Z077-90/pull/12

Once such PR is merged, I will create a new PR in the main repo to complete this issue.

mad-jrodriguez, 2023-01-03:
New PR created to finish the issue.

mad-jrodriguez, 2023-01-09:
Closing issue as the PR has been merged.

closed 2023-01-09 by mad-jrodriguez