MEN-Mikro-Elektronik/13MD05-90 #254: Build fails for Kernel >=5.14 (tried 5.15.76)

JFriesz, 2022-12-09:
The build breaks for kernel version >=5.14  

I noticed these deprecated calls / files during building for 5.15.76:

13Z025/men_z25_serial.c:33
`#include <linux/ide>`
=> linux/ide has been removed

Z001_SMB/z001_smb.c:130
`current->state = TASK_INTERRUPTIBLE;`
=> can probably be replaced by `WRITE_ONCE(current->__state, TASK_INTERRUPTIBLE);` for >=5.14

There might be more, as my Makefile only included the following:

```
ALL_LL_DRIVERS = \
	XM01BC/DRIVER/COM/driver.mak \
	SMB2/DRIVER/COM/driver.mak \
	MMODPRG/DRIVER/COM/driver.mak \
	MMODPRG/DRIVER/COM/driver_4k.mak \
	Z017/DRIVER/COM/driver.mak

ALL_BB_DRIVERS = \
	SMB2BB/DRIVER/NATIVE/driver.mak \
	CHAMELEON/DRIVER/COM/driver_pcitbl.mak

ALL_USR_LIBS = \
	USR_OSS/library.mak \
	USR_UTL/COM/library.mak \
	SMB2_API/COM/library.mak \
	SMB2_BMC_API/COM/library.mak \
	SMB2_SHC/COM/library.mak

ALL_CORE_LIBS = \
	DBG/library.mak \
	OSS/library.mak \
	CHAMELEON/COM/library.mak \

ALL_LL_TOOLS = \
	XM01BC/TOOLS/XM01BC_CTRL/COM/program.mak \
	SMB2/EXAMPLE/SMB2_SIMP/COM/program.mak \
	SMB2/EXAMPLE/SMB2_F601/COM/program.mak \
	SMB2/TOOLS/SMB2_CTRL/COM/program.mak \
	SMB2/TOOLS/SMB2_BOARDIDENT/COM/program.mak \
	SMB2/TOOLS/SMB2_TOUCH/COM/program.mak \
	SMB2/TOOLS/SMB2_BMC_CTRL/COM/program.mak \
	SMB2/TOOLS/SMB2_SHC_CTRL/COM/program.mak \
	SMB2/TOOLS/SMB2_STM32_FLASH/COM/program.mak \
	SMB2/TOOLS/SMB2_EETEMP/COM/program.mak \
	SMB2/TOOLS/SMB2_POE/COM/program.mak \
	Z017/EXAMPLE/Z17_SIMP/COM/program.mak \
	Z017/TOOLS/Z17_IO/COM/program.mak \
	Z017/TOOLS/Z127_IO/COM/program.mak \
	Z017/TOOLS/Z127_IN/COM/program.mak \
	Z017/TOOLS/Z127_OUT/COM/program.mak

ALL_COM_TOOLS = \
	MDIS_API/M_ERRSTR/COM/program.mak \
	MDIS_API/M_GETBLOCK/COM/program.mak \
	MDIS_API/M_GETSTAT/COM/program.mak \
	MDIS_API/M_GETSTAT_BLK/COM/program.mak \
	MDIS_API/M_MOD_ID/COM/program.mak \
	MDIS_API/M_OPEN/COM/program.mak \
	MDIS_API/M_READ/COM/program.mak \
	MDIS_API/M_REV_ID/COM/program.mak \
	MDIS_API/M_SETBLOCK/COM/program.mak \
	MDIS_API/M_SETSTAT/COM/program.mak \
	MDIS_API/M_SETSTAT_BLK/COM/program.mak \
	MDIS_API/M_WRITE/COM/program.mak \
	WDOG/WDOG_TEST/COM/program.mak \
	WDOG/WDOG_SIMP/COM/program.mak \

ALL_NATIVE_DRIVERS = \
	DRIVERS/Z001_SMB/driver_g2x.mak \
	DRIVERS/CHAMELEON/driver.mak \
	DRIVERS/13Z025/driver.mak \
	DRIVERS/Z069_RST_WDG/driver.mak

ALL_NATIVE_LIBS =

ALL_NATIVE_TOOLS =
```

dpfeuffer, 2022-12-09:
For the next 13MD05-90 release, "Yocto - Kirkstone - Kernel 5.15 64Bit (LTS)" support is demanded. If it is too much work, we may postpone this requirement.

mad-jsanjuan, 2022-12-13:
We actually have this compilation issue in more components. I've run a complete test for Linux 5.15 and all these failed:

```
jorge@jorge-Latitude-5530:~/Work/MDIS/mdis_test_system/MDIS_Compilation_Test/test/MDIS_Test/MDIS_Compilation/Results/2022_12_12_14_53_19/5.15_dbg$ cat TestResults.log | grep FAILED
Makefile.13Z055-90 FAILED
Makefile.13Z025-90 FAILED
Makefile.13M077-90 FAILED
Makefile.mdis_VME_16Z002 FAILED
Makefile.mdis_VME_16Z002_tsi148 FAILED
Makefile.13Z001-90 FAILED
```
[Makefile.13Z055-90.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/10216946/Makefile.13Z055-90.log)
[Makefile.13Z025-90.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/10216949/Makefile.13Z025-90.log)
[Makefile.13M077-90.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/10216953/Makefile.13M077-90.log)
[Makefile.mdis_VME_16Z002.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/102
[Makefile.mdis_VME_16Z002_tsi148.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/10216956/Makefile.mdis_VME_16Z002_tsi148.log)
16955/Makefile.mdis_VME_16Z002.log)
[Makefile.13Z001-90.log](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/10216959/Makefile.13Z001-90.log)

mad-jrodriguez, 2022-12-15:
I have fixed the compilation issue of M077.

From kernel version v5.13 the support for Cyclades devices was removed from the kernel tree. 

```
commit f76edd8f7ce06cdff2fe5b6b39a49644c684a161
Author: Jiri Slaby <jirislaby@kernel.org>
Date:   Tue Mar 2 07:21:35 2021 +0100

    tty: cyclades, remove this orphan
    
    The Cyclades driver was orphaned by commit d459883e6c54 (MAINTAINERS:
    remove two dead e-mail) 13 years ago. Noone stepped up to take care of
    them and to fix all the issues the driver has.
    
    On the top of that, there is no way to obtain the firmware for Z cards
    from the vendor as cyclades.com ceased to exist.
    
    So it's time to drop the driver with all its traces.
    
    Signed-off-by: Jiri Slaby <jslaby@suse.cz>
    Link: https://lore.kernel.org/r/20210302062214.29627-5-jslaby@suse.cz
    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>
```
The error was caused because the macro MEN_UART_MAJOR was not defined.

A Pull Request has been created to fix it.

https://github.com/MEN-Mikro-Elektronik/13M077-90/pull/1

mad-jrodriguez, 2022-12-15:
I have fixed the compilation issues from **VME_16Z002** and **VME_16Z002_tsi148**

The field was renamed in v5.14.

```
commit 2f064a59a11ff9bc22e52e9678bc601404c7cb34
Author: Peter Zijlstra <peterz@infradead.org>
Date:   Fri Jun 11 10:28:17 2021 +0200

    sched: Change task_struct::state
    
    Change the type and name of task_struct::state. Drop the volatile and
    shrink it to an 'unsigned int'. Rename it in order to find all uses
    such that we can use READ_ONCE/WRITE_ONCE as appropriate.
    
    Signed-off-by: Peter Zijlstra (Intel) <peterz@infradead.org>
    Reviewed-by: Daniel Bristot de Oliveira <bristot@redhat.com>
    Acked-by: Will Deacon <will@kernel.org>
    Acked-by: Daniel Thompson <daniel.thompson@linaro.org>
    Link: https://lore.kernel.org/r/20210611082838.550736351@infradead.org
```
Additionally, as it is mentioned in the commit message, the access to __state field is made using the macro READ_ONCE()

mad-jrodriguez, 2022-12-16:
I have fixed the compilation error for component:

- 13Z025-90

The support for legacy IDE was removed from kernel in v5.14.

```
commit b7fb14d3ac63117e0e8beabe75f4ea52051fbe3a
Author: Christoph Hellwig <hch@lst.de>
Date:   Wed Jun 16 15:46:58 2021 +0200
    ide: remove the legacy ide driver
    The legay ide driver has been replace with libata starting in 2003 and has
    been scheduled for removal for a while.  Finally kill it off so that we
    can start cleaning up various bits of cruft it forced on the block layer.
    Signed-off-by: Christoph Hellwig <hch@lst.de>
    Signed-off-by: Jens Axboe <axboe@kernel.dk>
```
Z025 never used the ide.h so we have removed it.

mad-jrodriguez, 2022-12-16:
We can consider this issue as finished so we will launch the compilation and functional testing.

closed 2022-12-21 by mad-jsanjuan