MEN-Mikro-Elektronik/13MD05-90 #257: QUICKSTART.md Instructions are inaccurate for RHEL-based systems

DaTurr3t, 2022-12-22:
Hello there

I recently tried to compile the Kernel Modules when a compilation error lead me to re-read the QUICKSTART.md. I noticed that the command "make prepare" is not needed for RHEL-based systems and only leads to confusion as it fails when following the instructions:

```
[root@devsystem ~]# yum install kernel-headers kernel-devel elfutils-libelf-devel
Package kernel-headers-3.10.0-1160.80.1.el7.x86_64 already installed and latest version
Package kernel-devel-3.10.0-1160.80.1.el7.x86_64 already installed and latest version
Package elfutils-libelf-devel-0.176-5.el7.x86_64 already installed and latest version
Nothing to do
[root@devsystem ~]# cd /usr/src/kernels/$(uname -r)
[root@devsystem 3.10.0-1160.80.1.el7.x86_64]# make oldconfig
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/basic/bin2c
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --oldconfig Kconfig
#
# configuration written to .config
#
[root@devsystem 3.10.0-1160.80.1.el7.x86_64]# make prepare
scripts/kconfig/conf --silentoldconfig Kconfig
make[1]: *** No rule to make target `arch/x86/syscalls/syscall_32.tbl', needed by `arch/x86/syscalls/../include/generated/uapi/asm/unistd_32.h'.  Stop.
make: *** [archheaders] Error 2
```

I first thought the compilation was unsuccessful because i have not "prepared" my sources, but unless one builds the Kernel Modules with cloned Kernel source code, this command is not needed. On RHEL 7 with Kernel 3.10, it is sufficient to install the packages and start the compilation of the modules.

Let me know if you need further information/clarification about this.

Thank you and kind regards
Matthias

dpfeuffer, 2023-01-02:
@DaTurr3t
Thanks for your input!

@mad-jsanjuan / @mad-jrodriguez 
Please check if the described problem exists also with official supported CentOS distros (e.g. 7.8, 8.2)
RHEL and kernel 3.10 is not official supported.

Nevertheless, the Quickstart Guide should not contain wrong instructions for:
"CentOs / RedHat
Note: RedHat Linux is not officially supported!"

So we have to fix the description or to remove the RedHat note.

mad-jsanjuan, 2023-01-03:
I have tried with a Virtual Machine with CentOS 8 (Kernel 4.18) and the Linux Kernel headers package is indeed prepared and ready to use. It contains the actual `.config`, `Module.symvers`, etc. There is no need to run any make commands inside the Kernel headers. 

These make commands would only be necessary if a raw **Linux Source** is used.

mad-jrodriguez, 2023-01-09:
Closing the issue has the PR has been merged

closed 2023-01-09 by mad-jrodriguez