MEN-Mikro-Elektronik/13MD05-90 #263: Provide 13Z055-90 "release" for testing in NBG

dpfeuffer, 2023-01-16:
Please merge the open pull requests in repo 13Z055-90
https://github.com/MEN-Mikro-Elektronik/13Z055-90/pulls
into mad-dev (see comments on the pull requests!).
Then run a compile test with the mad-dev branch.
If successful, merge mad-dev into master and mark the master commit with a release commit.
@DHofbauer 
After that we will pull release 13Z055-90 and DHofbauer will run operational tests with his test setup (for that he needs a release tag as reference for his tests).

mad-jrodriguez, 2023-01-16:
Hi @dpfeuffer 

I have approved and merged both PRs 

Additionally, I have run the compilation tests for Module 13Z055_90 and all have passed OK

```
4.9_dbg PASSED
4.9_nodbg PASSED
4.14_dbg PASSED
4.14_nodbg PASSED
4.19_dbg PASSED
4.19_nodbg PASSED
5.4_dbg PASSED
5.4_nodbg PASSED
5.10_dbg PASSED
5.10_nodbg PASSED
5.15_dbg PASSED
5.15_nodbg PASSED
```

Additionally, a PR have been created to merge the changes into the master branch

https://github.com/MEN-Mikro-Elektronik/13Z055-90/pull/8

Once the changes have been done in master, a new tag will be created with name: **13Z055-90_01_02**

mad-jsanjuan, 2023-01-17:
We have just merged all the development in `mad-dev` into `master` for the 13Z055-90 component. A new tag has been created https://github.com/MEN-Mikro-Elektronik/13Z055-90/releases/tag/13Z055-90_01_02

closed 2023-01-17 by mad-jsanjuan

dpfeuffer, 2023-01-17:
Thank you for your quick execution! 👍