MEN-Mikro-Elektronik/13MD05-90 #269: fpga_load cannot read io mapped chameleon table from F401

dpfeuffer, 2023-01-23:
The fpga_load tool should support to dump FPGA chameleon tables located at io mapped BAR.
With the F401 board (which I want to send to Spain), I was not able to read out the chameleon table.

HW:
F23P CPU, F401 

Linux:
PRETTY_NAME="CentOS Linux 8 (Core)"
Linux centos 4.18.0-80.11.2.el8_0.x86_64 #1 SMP Tue Sep 24 11:32:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```
[root@centos ~]# lspci -d 1172: -v
04:0e.0 Bridge: Altera Corporation Device 000a (rev 05)
        Subsystem: Device 4d45:5a14
        Flags: medium devsel, IRQ 255
        I/O ports at 4000 [size=4K]
        I/O ports at 5100 [size=256]
        I/O ports at 5000 [size=256]
        Memory at b0400000 (32-bit, non-prefetchable) [size=256]
        Expansion ROM at b0400800 [disabled] [size=2K]
```

I would assume to read out the chameleon table with one of these possibilities:

A)
```
[root@centos BIN]# ./fpga_load 0x1172 0x000a 0x4d45 0 -t -v
Z100_PCIInit

Update device:
Nr.|bus|dev|fun|   BAR0  |   BAR1  |   BAR2  |   BAR3  |   BAR4  |   BAR5
  0   4  14   0  00004001  00005101  00005001  b0400000  00000000  00000000
*** ERROR: No chameleon table found for PCI device (4/14/0) (bus/dev/fun) (0xc9)
*** ERROR: can't find device 0x02d
*** ERROR: Chameleon table or flash interface not found (0x2018)
```

B)
```
[root@centos BIN]# ./fpga_load -k 4001 -t -v
Z100_IsaInit
*** ERROR: no chameleon table found at address 0x4000 (0xc9)
*** ERROR: can't find device 0x02d
*** ERROR: Chameleon table or flash interface not found (0x2018)
```

I was able to dump the chameleon table with the help of the chameleon driver, so the chameleon table is present and can be read out with the chameleon library.
Here is my MDIS configuration for F23P + F401 + Z51_DAC:
[f401.tar.gz](https://github.com/MEN-Mikro-Elektronik/13MD05-90/files/10480936/f401.tar.gz)

Chameleon table dumped with the chameleon BBIS:
```
[root@centos work]# modprobe men_mdis_kernel
[root@centos work]# m_open dac_1
[root@centos work]# dmesg
..
[ 5269.792239] HCheckTable: V0 table found at tblAddr=d1bdd8b9
[ 5269.792268]  Unit                modCode Rev    Inst  IRQ  BAR Offset   Addr
[ 5269.792275]  00 16Z054_SYSTEM    0x0023  6   0x00    0x00   0   0x0000   0x00000000d1bdd8b9
[ 5269.792280]  01 16Z059_IRQ       0x0026  0   0x00    0x01   0   0x0100   0x00000000d37b0757
[ 5269.792286]  02 16Z049_TMR       0x001d  0   0x00    0x02   0   0x0200   0x00000000cc693d43
[ 5269.792291]  03 16Z051_DAC       0x0027  0   0x00    0x03   0   0x0300   0x000000008d9bce8d
[ 5269.792296]  04 16Z059_IRQ       0x0026  0   0x01    0x04   0   0x0400   0x0000000094a65d7f
[ 5269.792301]  05 ?                0x003e  0   0x00    0x08   0   0x0500   0x000000005a50d34d
[ 5269.792306]  06 16Z050_BIO       0x001e  0   0x00    0x05   1   0x0000   0x00000000721d7da8
[ 5269.792311]  07 16Z050_BIO       0x001e  0   0x01    0x06   2   0x0000   0x00000000fa58a764
[ 5269.792317]  08 16Z045_FLASH     0x001f  1   0x00    0x07   3   0x0000   0x0000000033640a70
```

dpfeuffer, 2023-03-17:
It looks good!

>The IP Core number 7 is not detected because the modCode is not defined. Do you know which kind of IP Core it it?

Don't know if I understand your question correctly. IP core 7 is listed:
`7 0x0032 16Z050_BIO                 0    1   0   0   6   2 0x00000000 0xe001`

Or do you mean scan_system.sh is not able to detect it?
Z50 is a GPIO device: https://github.com/MEN-Mikro-Elektronik/13Z050-06

dpfeuffer, 2023-03-17:
According to the F401 Spec, the IP core 5 is a debouncer:
[image]

This debouncer is reported with DevId 0xffff, because 0x3E DeviceID is unknown in chameleon.h.

The chameleon debugs should look like (see above for the full dump): 
`[ 5269.792301]  05 ?                0x003e  0   0x00    0x08   0   0x0500   0x000000005a50d34d`

But anyway, there is no MDIS driver for this debouncer. It is used to debounce the inputs of the 16Z050_BIO. The debouncer is configurable but it seems to me that it was never supported from a driver and only default values are used for debouncing.
The modCode 0x3E is not official defined in our FPGA IP Cor Overview document, so just let this IP core undefined.

mad-jrodriguez, 2023-03-17:
Hi Dieter,

Ok, so I will leave such IP core as "undefined". Then, we have to update the scan_system script in order to avoid such kind of lines, otherwise the script will fail.

Thank you so much.

mad-jrodriguez, 2023-03-21:
Hi @dpfeuffer 

After the rework, the patches are ready. Now, the Chameleon table is dumped with correct values, even for the table version and revision

```
men@men-MEN-F026L00:/opt/menlinux/TOOLS/FPGA_LOAD$ sudo ./fpga_load 0x1172 0x000a 0x4d45 0 -t -v
Z100_PCIInit

Update device:
Nr.|bus|dev|fun|   BAR0  |   BAR1  |   BAR2  |   BAR3  |   BAR4  |   BAR5
  0  14  14   0  0000d001  0000e101  0000e001  91601000  00000000  00000000

BARs of FPGA PCI device 14:14.0:
  BAR0: 0x0000d000; size: 0x00000000, mapType: IO;
  BAR1: 0x0000e100; size: 0x00000000, mapType: IO;
  BAR2: 0x0000e000; size: 0x00000000, mapType: IO;
  BAR3: 0x91601000; size: 0x00000000, mapType: MEM;
  BAR4: 0x00000000; size: 0x00000000, mapType: unused;
  BAR5: 0x00000000; size: 0x00000000, mapType: unused;

Information about the Chameleon FPGA:
FPGA File='         $  ' table model=0x41('A') Revision 5.0 Magic 0xABCD
List of the Chameleon units:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x0036 16Z054_SYSTEM              0    0   0   6   0   0 0x00000000 0xd000
  1 0x003b 16Z059_IRQ                 0    0   0   0   1   0 0x00000100 0xd100
  2 0x0031 16Z049_TMR                 0    0   0   0   2   0 0x00000200 0xd200
  3 0x0033 16Z051_DAC                 0    0   0   0   3   0 0x00000300 0xd300
  4 0x003b 16Z059_IRQ                 0    1   0   0   4   0 0x00000400 0xd400
  5 0xffff not identified             0    0   0   0   8   0 0x00000500 0xd500
  6 0x0032 16Z050_BIO                 0    0   0   0   5   1 0x00000000 0xe100
  7 0x0032 16Z050_BIO                 0    1   0   0   6   2 0x00000000 0xe000
  8 0x002d 16Z045_FLASH               0    0   0   1   7   3 0x00000000 0x91601000
```

mad-jsanjuan, 2023-03-31:
All work for this have been merged into mad-dev branch.

closed 2023-03-31 by mad-jsanjuan