MEN-Mikro-Elektronik/13MD05-90 #273: MDIS installation in Red Hat 9 is unusable

mad-jsanjuan, 2023-02-07:
When installing MDIS in Red Hat versio 9, the installed binaries and libraries are not being tracked by the system. The PATH environment variable and `ldconfig` will not let the user use the installed tools.

From the MDIS makefiles we get:

```
# The directory in which the shared (.so) user mode libraries
# are to be installed. Often something like /usr/local/lib.
# (relative to the target's root tree)

LIB_INSTALL_DIR = /usr/local/lib

# The directory in which the static user mode libraries are to
# be installed. Often something like /usr/local/lib on
# development host. For cross compilation select a path
# relative to your cross compilers lib directory.

STATIC_LIB_INSTALL_DIR = /usr/local/lib
```

```
[root@G25-MEN RedHat514070301el90x8664x8664]# echo $PATH
/root/.local/bin:/root/bin:/sbin:/bin:/usr/sbin:/usr/bin
[root@G25-MEN RedHat514070301el90x8664x8664]# ldd /usr/local/bin/m36_simp 
	linux-vdso.so.1 (0x00007ffda51e6000)
	libm.so.6 => /lib64/libm.so.6 (0x00007fc948417000)
	libmdis_api.so => not found
	libusr_oss.so => not found
	libusr_utl.so => not found
	libc.so.6 => /lib64/libc.so.6 (0x00007fc94820d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fc9484ff000)
[root@G25-MEN RedHat514070301el90x8664x8664]# ls -l /usr/local/lib/libmdis_api.so
-rw-r--r--. 1 root root 28520 Feb  7 05:38 /usr/local/lib/libmdis_api.so
```

We need to configure/document how to setup PATH and `ldconfig` stuff for Red Hat distros.

mad-jsanjuan, 2023-02-07:
This is only an issue when using the `scan_system.sh` script. The MDIS wizzard does let you choose these options, but this should probably also be supported on the `scan_system.sh` script.

dpfeuffer, 2023-02-08:
>We need to configure/document how to setup PATH and ldconfig stuff for Red Hat distros.
Yes, or that the user has to change the *_INSTALL_DIR within MDISWIZ.
Both should be described in QUICKSTART.md.

For the scan_system.sh, we could detect the running distro and change the paths automatically at runtime. What do you think?

dpfeuffer, 2023-02-08:
Ok, now I see in your pull request https://github.com/MEN-Mikro-Elektronik/13MD05-90/pull/274 that you already had the same idea. 👍

dpfeuffer, 2023-02-08:
In QUICKSTART.md is noted where the MDIS executables are located (see Binary Locations). Please adapt it accordingly. And I have to note this in the UM as well.

mad-jrodriguez, 2023-03-09:
Closing the issue as it is already merged with all changes suggested by @dpfeuffer.

closed 2023-03-09 by mad-jrodriguez