MEN-Mikro-Elektronik/13MD05-90 #276: mscan_loopb does not work

mad-jsanjuan, 2023-02-16:
The `mscan_loopb` program does not seem to work on a CAN interface:

https://github.com/MEN-Mikro-Elektronik/13Z015-06/blob/master/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/

```
$ sudo modprobe -r men_ll_z15
$ sudo modprobe men_ll_z15
$ sudo mscan_loopb -t=a can_1
=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===

*** Error during: mscan_read_msg( path, rxObj, 1000, &rxFrm ) == 0
file /opt/menlinux/DRIVERS/MDIS_LL/MSCAN/TOOLS/MSCAN_LOOPB/COM/mscan_loopb.c
line 412
ERROR (MDIS) 0x0802:  OSS: timeout occurred
Test a: FAILED
------------------------------------------------
TEST RESULT: 1 errors
Max irqtime=0 (internal ticks)
```

This was detected using the IP core z029 on a BL51 box pc.

dpfeuffer, 2023-02-20:
I talked with the FPGA developer of the 16Z029_MSCAN IP core.

Please try the following:
1) Test with 125kbit bitrate:
 `$ sudo mscan_loopb -t=a can_1 -b=4`
2) If it 1) doesn't work, provide us register dumps of the Z029 IP core
a) after power-up, before the men_ll_z15 is loaded
b) after the men_ll_z15 is loaded
c) after mscan_loopb execution

mad-jsanjuan, 2023-02-21:
Here is the output of @dpfeuffer suggestion:

Right After reboot:
```
men@men-BL51:/opt/menlinux/TOOLS/HWBUG$ sudo /opt/menlinux/BIN/fpga_load -k 0xf000e000 -t

Chameleon FPGA table for device 0xf000e000:
  BAR0: 0x00000000; size: 0x00001000, mapType: IO;
  BAR1: 0xf0000000; size: 0x00010000, mapType: MEM;
  BAR2: 0x00000000; size: 0x00000000, mapType: unused;
  BAR3: 0x00000000; size: 0x00000000, mapType: unused;
  BAR4: 0x00000000; size: 0x00000000, mapType: unused;
  BAR5: 0x00000000; size: 0x00000000, mapType: unused;

Information about the Chameleon FPGA:
FPGA File=' 16SC31-00A0' table model=0x41('A') Revision 0.0 Magic 0xABCE
List of the Chameleon units:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x007d 16Z125_UART                0    4   0  17   6   0 0x00000220 0x220
  1 0x007d 16Z125_UART                0    0   0  17   4   0 0x000003f8 0x3f8
  2 0x007d 16Z125_UART                0    1   0  17   3   0 0x000002f8 0x2f8
  3 0x007d 16Z125_UART                0    2   0  17   7   0 0x000003e8 0x3e8
  4 0x007d 16Z125_UART                0    3   0  17   5   0 0x000002e8 0x2e8
  5 0x0052 16Z082_IMPULSE             0    0   0   2  63   1 0x00000800 0xf0000800
  6 0x0022 16Z034_GPIO                0    0   0  10  11   1 0x0000e200 0xf000e200
  7 0x0022 16Z034_GPIO                0    2   0  10  11   1 0x0000e240 0xf000e240
  8 0x0025 16Z037_GPIO                0    0   1   1  11   1 0x0000e260 0xf000e260
  9 0x007e 16Z126_SERFLASH            0    0   1   6  63   1 0x0000e280 0xf000e280
 10 0x001d 16Z029_CAN                 0    0   0  18  11   1 0x0000e400 0xf000e400
 11 0x004c 16Z076_QSPI                0    0   0   7  11   1 0x0000e800 0xf000e800
 12 0x001d 16Z029_CAN                 0    1   0  18  11   1 0x0000f000 0xf000f000

 Current FPGA file/usage status: ** invalid, Check FPGA programming.

men@men-BL51:/opt/menlinux/TOOLS/HWBUG$ sudo ./hwbug_cmd 0xf000e400 0xf000e4ff -d
Dump memory f000e400..f000e4ff
f000e400+00000000: 01 00 00 00 81 00 00 00 01 00 00 00 13 00 00 00 ................
f000e400+00000010: 00 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 ................
f000e400+00000020: 00 00 00 00 4d 00 00 00 00 00 00 00 00 00 00 00 ....M...........
f000e400+00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ...............
```

Modprobing the linux driver:

```
men@men-BL51:/opt/menlinux/TOOLS/HWBUG$ sudo modprobe men_ll_z15
men@men-BL51:/opt/menlinux/TOOLS/HWBUG$ sudo ./hwbug_cmd 0xf000e400 0xf000e4ff -d
Dump memory f000e400..f000e4ff
f000e400+00000000: 01 00 00 00 81 00 00 00 01 00 00 00 13 00 00 00 ................
f000e400+00000010: 00 00 00 00 00 00 00 00 07 00 00 00 00 00 00 00 ................
f000e400+00000020: 00 00 00 00 4d 00 00 00 00 00 00 00 00 00 00 00 ....M...........
f000e400+00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000050: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000060: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
[... 16 lines cut ...]
line 412
ERROR (MDIS) 0x0802:  OSS: timeout occurred
Test a: FAILED
------------------------------------------------
TEST RESULT: 1 errors
Max irqtime=0 (internal ticks)
men@men-BL51:/opt/menlinux/TOOLS/HWBUG$ sudo ./hwbug_cmd 0xf000e400 0xf000e4ff -d
Dump memory f000e400..f000e4ff
f000e400+00000000: 01 00 00 00 a1 00 00 00 0f 00 00 00 1c 00 00 00 ................
f000e400+00000010: 00 00 00 00 3c 00 00 00 07 00 00 00 00 00 00 00 ....<...........
f000e400+00000020: 00 00 00 00 4d 00 00 00 00 00 00 00 00 00 00 00 ....M...........
f000e400+00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000050: 00 00 00 00 18 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+00000060: ff 00 00 00 f7 00 00 00 ff 00 00 00 ff 00 00 00 ................
f000e400+00000070: ff 00 00 00 e7 00 00 00 ff 00 00 00 ff 00 00 00 ................
f000e400+00000080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
[... cut ...]
f000e400+000000a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000e0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ................
f000e400+000000f0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ...............
```

dpfeuffer, 2023-02-22:
Hi Jorge,

I answer via mail because we are currently in island mode 😉

According our FPGA developer, the register dumps lock reasonable.
Please investigate if the interrupts from FPGA to Linux works.
You can check if Linux gets some interrupts via command line.

Or try to trigger some interrupts at one of the GPIO units which shares the same IRQ line:

  6 0x0022 16Z034_GPIO                0    0   0  10  11   1 0x0000e200 0xf000e200

  7 0x0022 16Z034_GPIO                0    2   0  10  11   1 0x0000e240 0xf000e240

  8 0x0025 16Z037_GPIO                0    0   1   1  11   1 0x0000e260 0xf000e260

 10 0x001d 16Z029_CAN                 0    0   0  18  11   1 0x0000e400 0xf000e400

 11 0x004c 16Z076_QSPI                0    0   0   7  11   1 0x0000e800 0xf000e800

 12 0x001d 16Z029_CAN                 0    1   0  18  11   1 0x0000f000 0xf000f000

---
Regards
Dieter

mad-jsanjuan, 2023-03-06:
I found this Errata on the BL51 box pc manual (https://www.duagon.com/products/details/BL51E/#doc) :

[image]

Is it possible this example program was never meant to run on this BL51?

mad-jrodriguez, 2023-03-07:
Hi all,

I have tested mscan_loopb executable in CAN buses within F215 and G215 boards and they work as expected so the issue detected in BL51E seems to be related with the Errata @mad-jsanjuan found yesterday.

CAN_7 G215
```
men@men-F027:~$ sudo mscan_loopb -t=a can_7
=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===
Test a: ok
------------------------------------------------
TEST RESULT: 0 errors
Max irqtime=0 (internal ticks)
```

CAN_8 G215
```
men@men-F027:~$ sudo mscan_loopb -t=a can_8
=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===
Test a: ok
------------------------------------------------
TEST RESULT: 0 errors
Max irqtime=0 (internal ticks)
```

CAN_15 F215
```
men@men-F027:~$ sudo mscan_loopb -t=a can_15
=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===
Test a: ok
------------------------------------------------
TEST RESULT: 0 errors
Max irqtime=0 (internal ticks)
```

CAN_16 F215
```
men@men-F027:~$ sudo mscan_loopb -t=a can_16
=== Performing test a: Basic Tx/Rx                                 (Run 1/1) ===
Test a: ok
------------------------------------------------
TEST RESULT: 0 errors
Max irqtime=0 (internal ticks)
```

dpfeuffer, 2023-03-13:
Good work! I wasn't aware of the mentioned BL51 errata and the known CAN issues. I think you are right and I close this issue now.

closed 2023-03-13 by dpfeuffer