MEN-Mikro-Elektronik/13MD05-90 #277: Build fails while trying to locate stdarg.h: fatal error: stdarg.h: No such file or directory

DaTurr3t, 2023-02-17:
Hello there!

I thought this had already been fixed on the mad-dev branch, but i had this error again on the latest commit (as of writing) of mad-dev.
Whilst compiling MDIS5 for Red Hat Enterprise Linux 9 (Kernel 5.14.0-162.6.1.el9_1.x86_64), i encountered the compiler complaining that it was unable to find stdarg.h.
Since at least kernel 5.14, stdarg.h has been moved from `/usr/src/linux/include/stdarg.h` to `/usr/src/linux/include/linux/stdarg.h`.

Modifying `INCLUDE/NATIVE/MEN/oss_os.h` to use `linux/stdarg.h` instead fails a bit further down the log with a similar error, probably due to different Include Paths.

The workaround I use at the moment is to symlink the old location to the new one: `ln -s /usr/src/kernels/$(uname -r)/include/linux/stdarg.h /usr/src/kernels/$(uname -r)/include/stdarg.h`.

Let me know if you need any further information.

Compilation Error log:

```
+ make
Getting Compiler/Linker settings from Linux Kernel Makefile
Cleaning .kernelsubdirs

++++++++ Preparing non-debug version of module men_mdis_kernel +++++++++++
Directory OBJ/nodbg/men_mdis_kernel created

++++++++ Preparing non-debug version of module men_bbis_kernel +++++++++++
Directory OBJ/nodbg/men_bbis_kernel created

++++++++ Preparing non-debug version of module men_oss +++++++++++
Directory OBJ/nodbg/men_oss created

++++++++ Preparing non-debug version of module men_dbg +++++++++++
Directory OBJ/nodbg/men_dbg created

++++++++ Preparing non-debug version of module men_desc +++++++++++
Directory OBJ/nodbg/men_desc created

++++++++ Preparing non-debug version of module men_pld +++++++++++
Directory OBJ/nodbg/men_pld created

++++++++ Preparing non-debug version of module men_pld_sw +++++++++++
Directory OBJ/nodbg/men_pld_sw created

++++++++ Preparing non-debug version of module men_id +++++++++++
Directory OBJ/nodbg/men_id created

++++++++ Preparing non-debug version of module men_id_sw +++++++++++
Directory OBJ/nodbg/men_id_sw created

++++++++ Preparing non-debug version of module men_mbuf +++++++++++
Directory OBJ/nodbg/men_mbuf created

++++++++ Preparing non-debug version of module men_smb_port +++++++++++
Directory OBJ/nodbg/men_smb_port created

++++++++ Preparing non-debug version of module men_smb_port_io +++++++++++
Directory OBJ/nodbg/men_smb_port_io created

++++++++ Preparing non-debug version of module men_chameleon +++++++++++
Directory OBJ/nodbg/men_chameleon created

++++++++ Preparing non-debug version of module men_chameleon_io +++++++++++
Directory OBJ/nodbg/men_chameleon_io created

++++++++ Preparing non-debug version of module men_ll_z17 +++++++++++
Directory OBJ/nodbg/men_ll_z17 created

++++++++ Preparing non-debug version of module men_ll_xm01bc +++++++++++
Directory OBJ/nodbg/men_ll_xm01bc created

++++++++ Preparing non-debug version of module men_ll_smb2 +++++++++++
Directory OBJ/nodbg/men_ll_smb2 created

++++++++ Preparing non-debug version of module men_ll_z15 +++++++++++
Directory OBJ/nodbg/men_ll_z15 created

++++++++ Preparing non-debug version of module men_ll_z82 +++++++++++
Directory OBJ/nodbg/men_ll_z82 created

++++++++ Preparing non-debug version of module men_bb_chameleon_custom +++++++++++
Directory OBJ/nodbg/men_bb_chameleon_custom created

++++++++ Preparing non-debug version of module men_bb_smb2 +++++++++++
Directory OBJ/nodbg/men_bb_smb2 created
++++++++ Building kernel modules ++++++++++
cat /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/.kernelsubdirs >>OBJ/Makefile
### Compiling kernel modules ###
  CC [M]  /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/OBJ/nodbg/men_mdis_kernel/mk_module.o
In file included from /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/INCLUDE/COM/MEN/oss.h:39,
                 from /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/LIBSRC/MDIS_KERNEL/mk_intern.h:52,
                 from /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/OBJ/nodbg/men_mdis_kernel/mk_module.c:31:
/home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/INCLUDE/NATIVE/MEN/oss_os.h:36:10: fatal error: stdarg.h: No such file or directory
   36 | #include <stdarg.h> /* for va_list */
      |          ^~~~~~~~~~
compilation terminated.
make[4]: *** [scripts/Makefile.build:295: /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/OBJ/nodbg/men_mdis_kernel/mk_module.o] Error 1
make[3]: *** [scripts/Makefile.build:574: /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/OBJ/nodbg/men_mdis_kernel] Error 2
make[2]: *** [Makefile:1915: /home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/OBJ] Error 2
make[1]: *** [/home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/BUILD/MDIS/TPL/rules.mak:503: callkernelbuild] Error 2
make: *** [/home/daturr3t/projects/wowbagger-rpm-mdis/rpmbuild/BUILD/custom-mdis5-5.14.0/mdis5-custom/BUILD/MDIS/TPL/rules.mak:401: nodbg] Error 2
```

I added the following components to my makefile:
```
ALL_LL_DRIVERS = \
        Z017/DRIVER/COM/driver.mak \
        XM01BC/DRIVER/COM/driver.mak \
        SMB2/DRIVER/COM/driver.mak \
        MSCAN/DRIVER/COM/driver_z15.mak \
        Z082/DRIVER/COM/driver.mak

ALL_BB_DRIVERS = CHAMELEON/DRIVER/COM/driver_custom.mak \
                 SMB2BB/DRIVER/NATIVE/driver.mak

ALL_USR_LIBS = \
        USR_OSS/library.mak \
        USR_UTL/COM/library.mak \
        MSCAN_API/COM/library.mak \
        SMB2_API/COM/library.mak \
        SMB2_SHC/COM/library.mak

ALL_CORE_LIBS = \
        DBG/library.mak \
        OSS/library.mak \
        CHAMELEON/COM/library.mak

ALL_LL_TOOLS = \
        Z017/EXAMPLE/Z17_SIMP/COM/program.mak \
        Z017/TOOLS/Z17_IO/COM/program.mak \
        Z017/TOOLS/Z127_IO/COM/program.mak \
        MSCAN/TOOLS/MSCAN_ALYZER/COM/program.mak \
        MSCAN/TOOLS/MSCAN_LOOPB/COM/program.mak \
        MSCAN/TOOLS/MSCAN_MENU/COM/program.mak \
        SMB2/EXAMPLE/SMB2_SIMP/COM/program.mak \
        SMB2/EXAMPLE/SMB2_F601/COM/program.mak \
        SMB2/TOOLS/SMB2_CTRL/COM/program.mak \
        SMB2/TOOLS/SMB2_BOARDIDENT/COM/program.mak \
        SMB2/TOOLS/SMB2_TOUCH/COM/program.mak \
        SMB2/TOOLS/SMB2_BMC/COM/program.mak \
        SMB2/TOOLS/SMB2_SHC_CTRL/COM/program.mak \
        SMB2/TOOLS/SMB2_STM32_FLASH/COM/program.mak \
        SMB2/TOOLS/SMB2_EETEMP/COM/program.mak \
        SMB2/TOOLS/SMB2_POE/COM/program.mak \
        Z082/EXAMPLE/Z82_SIMP/COM/program.mak \
        XM01BC/TOOLS/XM01BC_CTRL/COM/program.mak
ALL_COM_TOOLS = \
        MDIS_API/M_ERRSTR/COM/program.mak \
        MDIS_API/M_GETBLOCK/COM/program.mak \
        MDIS_API/M_GETSTAT/COM/program.mak \
        MDIS_API/M_GETSTAT_BLK/COM/program.mak \
        MDIS_API/M_MOD_ID/COM/program.mak \
        MDIS_API/M_OPEN/COM/program.mak \
        MDIS_API/M_READ/COM/program.mak \
        MDIS_API/M_REV_ID/COM/program.mak \
        MDIS_API/M_SETBLOCK/COM/program.mak \
        MDIS_API/M_SETSTAT/COM/program.mak \
        MDIS_API/M_SETSTAT_BLK/COM/program.mak \
        MDIS_API/M_WRITE/COM/program.mak

ALL_NATIVE_DRIVERS =

ALL_NATIVE_LIBS =

ALL_NATIVE_TOOLS =
```

Thank you and kind regards
Matthias

dpfeuffer, 2023-02-17:
@mad-jsanjuan 
@mad-jrodriguez 
Please try to reproduce the described problem with RHEL 9.1 kernel 5.15 which we will official support with the next release.

mad-jsanjuan, 2023-02-17:
Thanks for reporting this. We can reproduce now in our HW setups. We did not detect this before because our Red Hat system had `5.14.0-70.30.1.el9_0.x86_64` kernel package installed. I just updated the kernel and with this package we get the same error. 

```
[men@G25-MEN RedHat5140162121el91x8664x8664]$ cd /usr/src/kernels/
[men@G25-MEN kernels]$ ls -al
total 8
drwxr-xr-x.  4 root root   77 Feb  1 09:21 .
drwxr-xr-x.  5 root root   62 Feb 17 06:49 ..
drwxr-xr-x. 24 root root 4096 Feb  1 09:21 5.14.0-162.12.1.el9_1.x86_64
drwxr-xr-x. 24 root root 4096 Feb 17 06:46 5.14.0-70.30.1.el9_0.x86_64
[men@G25-MEN kernels]$ find . -name stdarg.h
./5.14.0-162.12.1.el9_1.x86_64/include/linux/stdarg.h
```

It looks `stdarg.sh` did not even exist in the older package. Is it possible our MDIS Makefiles were picking it from /usr/include? 

Probably the best way to try to fix this is from our Makefiles. We will investigate and fix this.

mad-jsanjuan, 2023-02-20:
I did some research and I think I found the actual problem. The file stdarg.h file did never exist in the Kernel sources until 5.15. Instead, whatever the toolchain includes provided was being used when doing `#include <stdarg.h>`. Eg:

```
$ find /usr/ -name stdarg.h
/usr/lib/gcc/x86_64-redhat-linux/11/include/stdarg.h
```

Now, the kernel sources ship a minimal version of stdarg.h and this should be used when compiling in kernel space. So we will need include the correct header file for userspace and kernel space like so:

```C
#ifdef __KERNEL__
#include <linux/stdarg.h>
#else 
#include <stdarg.h>
#endif
```

This was introduced with this patch in the kernel tree: https://github.com/torvalds/linux/commit/c0891ac15f0428ffa81b2e818d416bdf3cb74ab6

mad-jsanjuan, 2023-02-20:
For reference, here is exactly where the kernel prevent accidental inclusion of undesireable headers:  https://github.com/torvalds/linux/commit/04e85bbf71c9072dcf0ad9a7150495d72461105c

So from kernel v5.16 is really where the `#include <stdarg.h>` directive will stop working as it would be including things from the compilation toolchain and not the kernel sources.