MEN-Mikro-Elektronik/13MD05-90 #281: Compilation in kernel 6.1 is broken

mad-jsanjuan, 2023-02-22:
There is something in 6.1 that has changed the way some of the defined guards for VXWORK work. We get this error:

```
  CC [M]  /home/docker/mdis_test_system/actions-runner/_work/13MD05-90/13MD05-90/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon/chameleonv2.o
/home/docker/mdis_test_system/actions-runner/_work/13MD05-90/13MD05-90/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon/chameleonv2.c:270:5: error: "_WRS_VXWORKS_MAJOR" is not defined, evaluates to 0 [-Werror=undef]
  270 | #if _WRS_VXWORKS_MAJOR == 7
      |     ^~~~~~~~~~~~~~~~~~
/home/docker/mdis_test_system/actions-runner/_work/13MD05-90/13MD05-90/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon/chameleonv2.c: In function 'InitPci':
/home/docker/mdis_test_system/actions-runner/_work/13MD05-90/13MD05-90/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon/chameleonv2.c:409:5: error: "_WRS_VXWORKS_MAJOR" is not defined, evaluates to 0 [-Werror=undef]
  409 | #if _WRS_VXWORKS_MAJOR == 7
      |     ^~~~~~~~~~~~~~~~~~
/home/docker/mdis_test_system/actions-runner/_work/13MD05-90/13MD05-90/mdis_test_system/MDIS_Compilation_Test/OBJ/dbg/men_chameleon/chameleonv2.c:515:5: error: "_WRS_VXWORKS_MAJOR" is not defined, evaluates to 0 [-Werror=undef]
  515 | #if _WRS_VXWORKS_MAJOR == 7
      |     ^~~~~~~~~~~~~~~~~~
```

We need to fix this and also find why this only affects 6.1 and not older versions.

mad-jsanjuan, 2023-03-09:
Almost fixed. Just missing this one bit: https://github.com/MEN-Mikro-Elektronik/13MD05-90/issues/283

mad-jrodriguez, 2023-03-09:
Closing the issue. Further discussions should be done in #283 or in a new ticket.

closed 2023-03-09 by mad-jrodriguez