MEN-Mikro-Elektronik/13MD05-90 #283: virt_to_bus() deprecated from Linux Kernel v6.1

mad-jsanjuan, 2023-03-02:
The virt_to_bus()/virt_to_bus()  functions have been deprecated in the kernel https://github.com/torvalds/linux/commit/4313a24985f00340eeb591fd66aa2b257b9e0a69

MDIS uses virt_to_bus() in a couple places to implement DMA drivers: 

1) Tundra TSI148 vme bridge uses the vme4l-tsi148 driver. This is needed for controlling VME on CPU processor A17, A19 and A20. 
2) MEN Z76 QSPI controller (https://github.com/MEN-Mikro-Elektronik/13Z010-06). The z62 DMA implementation (dma_z62.c) makes use of virt_to_bus(). 

The errors we get for those are:

```
error: implicit declaration of function 'virt_to_bus'; did you mean 'virt_to_fix'? [-Werror=implicit-function-declaration]
 1897 |         tmp64 = virt_to_bus(bdVirtP);
```

This change in the kernel does not seem like a simple api change but more of a complete deprecation of some implementation so the change may not be so simple. Also, verifying any changes for this DMA implementations may need some planning. 

One initial solution around this could be to make the preprocessor fail for kernel versions greater than 6.1 if someone attempts to compile these modules. They seem to be already in End of Support status already but we will have to confirm this first.

NOTE: For reference, there is this patch in the kernel from someone that has dome something that may be similar: https://github.com/torvalds/linux/commit/a6958951ebe7db60e84b2437ee53aa4843028726

mad-jrodriguez, 2023-03-13:
Already fixed. The issue will be fixed once the associated issue https://github.com/MEN-Mikro-Elektronik/mdis_test_system/issues/49 is closed.

mad-jsanjuan, 2023-03-15:
The "solution" for this issue was to just do a controlled failure in the compilation of software components that use `virt_to_bus()` with Linux Kernel Version greater than v6.1 . Given this virt_to_bus thing has been deprecated, the linux kernel community have been deprecating all drivers using them. 

This issue may show up in the future if we are to keep supporting the DMA implementations in MDIS that use this for newer kernels. This will require a development plan on how to do it and how to test this.

dpfeuffer, 2023-03-16:
From the current PM requirements (13MD05-06_MDIS_SupportMatrix.xlsm 0.36):
- A17: not listed
- A19, A20: EOS
- 16Z076_QSPI: EOS

EOS: Legacy product. Product support is still available, but not functionally tested any more during regression tests. 100% functionality can not be guaranteed.

Therefore, it should be ok to skip support for Linux Kernel Version greater than v6.1.

closed 2023-03-22 by mad-jrodriguez