MEN-Mikro-Elektronik/13MD05-90 #292: Update scan_system script to avoid parsing "not identified" IP Cores

mad-jrodriguez, 2023-03-17:
During the dump of the Chameleon table of board F401 we have detected one of the IP Cores are not identified. As we discussed in #269 such IP core should not be supported so we have to skip such line in order to avoid possible issues while running the script.

```
Information about the Chameleon FPGA:
FPGA File='         $  ' table model=0x00('') Revision 65.0 Magic 0xABCD
List of the Chameleon units:
Idx DevId  Module                   Grp Inst Var Rev IRQ BAR Offset     Address
--- ------ ------------------------ --- ---- --- --- --- --- ---------- ----------
  0 0x0036 16Z054_SYSTEM              0    0   0   6   0   0 0x00000000 0xd001
  1 0x003b 16Z059_IRQ                 0    0   0   0   1   0 0x00000100 0xd101
  2 0x0031 16Z049_TMR                 0    0   0   0   2   0 0x00000200 0xd201
  3 0x0033 16Z051_DAC                 0    0   0   0   3   0 0x00000300 0xd301
  4 0x003b 16Z059_IRQ                 0    1   0   0   4   0 0x00000400 0xd401
  5 0xffff not identified             0    0   0   0   8   0 0x00000500 0xd501
  6 0x0032 16Z050_BIO                 0    0   0   0   5   1 0x00000000 0xe101
  7 0x0032 16Z050_BIO                 0    1   0   0   6   2 0x00000000 0xe001
  8 0x002d 16Z045_FLASH               0    0   0   1   7   3 0x00000000 0x91601000
```

mad-jrodriguez, 2023-03-22:
After fix the fpga_load and the scan_system script, I've just run it and only 3 IP Cores are detected of 2 different kind of IP Cores.

```
mezz_cham_1 {

    # ------------------------------------------------------------------------
    #        general parameters (don't modify)
    # ------------------------------------------------------------------------
    DESC_TYPE = U_INT32 0x2
    HW_TYPE = STRING CHAMELEON_PCITBL
    _WIZ_MODEL = STRING MEZZ_CHAM
    _WIZ_BUSIF = STRING cpu,1

    # ------------------------------------------------------------------------
    #        PCI configuration
    # ------------------------------------------------------------------------
    PCI_BUS_NUMBER = U_INT32 0xe
    PCI_DEVICE_NUMBER = U_INT32 0xe

    # ------------------------------------------------------------------------
    #        Chameleon BBIS Device: 
    #            DEVICE_IDV2_X is:   ((Cham devId) << 8 | instance)
    #        inside groups:  
    #            DEVICE_IDV2_X is:   ((Cham devId) << 8 | index inside group )
    # ------------------------------------------------------------------------

    DEVICE_IDV2_0 = U_INT32 0x3300
    DEVICE_IDV2_1 = U_INT32 0x3200
    DEVICE_IDV2_2 = U_INT32 0x3201

    # ------------------------------------------------------------------------
    #        debug levels (optional)
    #        these keys have only effect on debug drivers
    # ------------------------------------------------------------------------
    DEBUG_LEVEL = U_INT32 0xc0008000
    DEBUG_LEVEL_BK = U_INT32 0xc0008000
    DEBUG_LEVEL_OSS = U_INT32 0xc0008000
    DEBUG_LEVEL_DESC = U_INT32 0xc0008000
}
dac_1 {

    # ------------------------------------------------------------------------
    #        general parameters (don't modify)
    # ------------------------------------------------------------------------
    DESC_TYPE = U_INT32 0x1
    HW_TYPE = STRING Z51
    _WIZ_MODEL = STRING 16Z051_DAC

    # ------------------------------------------------------------------------
    #        reference to base board
    # ------------------------------------------------------------------------
    BOARD_NAME = STRING mezz_cham_1
    DEVICE_SLOT = U_INT32 0x0

    # ------------------------------------------------------------------------
    #        device parameters
    # ------------------------------------------------------------------------
    
}
bioc_2 {

    # ------------------------------------------------------------------------
    #        general parameters (don't modify)
    # ------------------------------------------------------------------------
    DESC_TYPE = U_INT32 0x1
    HW_TYPE = STRING Z50
    _WIZ_MODEL = STRING 16Z050_BIO

    # ------------------------------------------------------------------------
    #        reference to base board
    # ------------------------------------------------------------------------
    BOARD_NAME = STRING mezz_cham_1
    DEVICE_SLOT = U_INT32 0x1

    # ------------------------------------------------------------------------
    #        device parameters
    # ------------------------------------------------------------------------
    
}
bioc_3 {

    # ------------------------------------------------------------------------
    #        general parameters (don't modify)
    # ------------------------------------------------------------------------
    DESC_TYPE = U_INT32 0x1
    HW_TYPE = STRING Z50
    _WIZ_MODEL = STRING 16Z050_BIO

    # ------------------------------------------------------------------------
    #        reference to base board
    # ------------------------------------------------------------------------
    BOARD_NAME = STRING mezz_cham_1
    DEVICE_SLOT = U_INT32 0x2

    # ------------------------------------------------------------------------
    #        device parameters
    # ------------------------------------------------------------------------
    
}
```

- 16Z051_DAC
- 16Z050_BIO

dpfeuffer, 2023-03-22:
After thinking about it, your result is ok. We have only LL driver for 16Z051_DAC and 16Z050_BIO but not for the other IP cores in the FPGA. Therefore, the scan_system "database" provides no match for the other implemented IP cores.

mad-jrodriguez, 2023-03-22:
Perfect, so I will leave as it is. I'm going to submit changes in scan_system in order to detect the F401 because the current filters don't match  the new board

closed 2023-03-23 by mad-jrodriguez