MEN-Mikro-Elektronik/13MD05-90 #294: Generated Makefile has broken includes

areitz86, 2023-03-24:
[image]

[image]

Generated Makefile from MDIS scan has wrong file extensions in last entry per block. Other entries have a continuation slash, even though there no next line.

areitz86, 2023-03-24:
Commit: **https://github.com/MEN-Mikro-Elektronik/13MD05-90/commit/5bb36c83baac4722b9649991c4ab06500d5e653e**

areitz86, 2023-03-28:
hey @mad-jsanjuan,

bash version is 4.2.46(2) (x86_64-redhatz-linux-gnu) on Scientific Linux (RHEL) 7.4.

Cheers,
Andreas

mad-jsanjuan, 2023-03-30:
@areitz86 . This Scientific Linux (RHEL 7.4) is not officially supported by MDIS 13MD05-90. Please check https://github.com/MEN-Mikro-Elektronik/13MD05-90/blob/mad-dev/supported_components.md although this still needs updating for the upcoming release. 

We have however looked into making the scan script a bit more flexible and it should not add this extra quotation marks anymore.

areitz86, 2023-03-30:
Thank you. I was aware that RHEL7.4 is not supported. However, my thought was, that the scanning is depending on standard utils like bash,sed,etc., it should not fail on old but standard userspace tools.
Thank you really much for looking into it. Will check and report back, but close the issue for now.

closed 2023-03-30 by areitz86