MEN-Mikro-Elektronik/13MD05-90 #297: Add updated binaries for release

mad-jsanjuan, 2023-04-03:
Before packing the release we need to update `mdiswiz` and `fpga_load` binaries binaries as they have both been updated for this 13MD05-90_02_05 release.

mad-jrodriguez, 2023-04-03:
Hi @dpfeuffer 

Reviewing the binaries to be updated, we have found there are 2 binaries for fpga_load

- fpga_load
- fpga_load_powerpc

shall we need to rebuild also fpga_load_powerpc? or is it already deprecated?

dpfeuffer, 2023-04-03:
PowerPC support is EOL.
Just build the fpga_load binary for x86 32-bit (this should run also at x64 64-bit CPU architectures).

mad-jrodriguez, 2023-04-14:
I have just updated the fpga_load binary

```
fpga_load: ELF 32-bit LSB executable, Intel 80386, version 1 (GNU/Linux), statically linked, for GNU/Linux 3.2.0, BuildID[sha1]=8fccb55936a18755bb9a14253d5c9be630bac108, stripped
```
It also works on my laptop which has x86_64 arch.

closed 2023-04-27 by mad-jsanjuan